gmacario/easy-build #94: build-openwrt: make feeds/packages/libs/intltool fails in configure

gmacario, 2015-01-12:
NOTE: local patches applied on openwrt master
(git-svn-id: svn://svn.openwrt.org/openwrt/trunk@43942 3c298f89-4303-0410-b956-a3cf2f4a3e73)

```
...
 make[3] -C feeds/packages/utils/nano compile
 make[3] -C feeds/packages/libs/expat compile
 make[3] -C feeds/packages/utils/dbus compile
 make[3] -C feeds/packages/libs/gdbm compile
 make[3] -C feeds/packages/libs/intltool host-compile
make -r world: build failed. Please re-run make with V=s to see what's going on
make: *** [world] Error 1
build@1c118a3b61d1:~/openwrt$
```

gmacario, 2015-01-12:
Seeing what's going on:
```
build@1c118a3b61d1:~/openwrt$ make V=s
make[1]: Entering directory `/home/build/openwrt'
make[2]: Entering directory `/home/build/openwrt'
make[3]: Entering directory `/home/build/openwrt/target/linux'
make[4]: Entering directory `/home/build/openwrt/target/linux/ramips'
/home/build/openwrt/scripts/kconfig.pl  + /home/build/openwrt/target/linux/generic/config-3.14 /home/build/openwrt/target/linux/ramips/mt7620/config-3.14 > /home/build/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7620/linux-3.14.28/.config.target
...
make[3]: Leaving directory `/home/build/openwrt/feeds/packages/libs/gdbm'
make[3]: Entering directory `/home/build/openwrt/feeds/packages/libs/intltool'
(cd /home/build/openwrt/build_dir/host/intltool-0.40.6/; if [ -x configure ]; then cp -fpR /home/build/openwrt/scripts/config.{guess,sub} /home/build/openwrt/build_dir/host/intltool-0.40.6// &&  /bin/bash ./configure CC="gcc" CFLAGS="-O2 -I/home/build/openwrt/staging_dir/host/include -I/home/build/openwrt/staging_dir/host/usr/include" CPPFLAGS="-I/home/build/openwrt/staging_dir/host/include -I/home/build/openwrt/staging_dir/host/usr/include" LDFLAGS="-L/home/build/openwrt/staging_dir/host/lib -L/home/build/openwrt/staging_dir/host/usr/lib" SHELL="/bin/bash" --target=x86_64-linux-gnu --host=x86_64-linux-gnu --build=x86_64-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/build/openwrt/staging_dir/host --exec-prefix=/home/build/openwrt/staging_dir/host --sysconfdir=/home/build/openwrt/staging_dir/host/etc --localstatedir=/home/build/openwrt/staging_dir/host/var --sbindir=/home/build/openwrt/staging_dir/host/bin ; fi )
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for perl... /usr/bin/perl
checking for perl >= 5.8.1... 5.18.2
checking for XML::Parser... configure: error: XML::Parser perl module is required for intltool
make[3]: *** [/home/build/openwrt/build_dir/host/intltool-0.40.6/.configured] Error 1
make[3]: Leaving directory `/home/build/openwrt/feeds/packages/libs/intltool'
make[2]: *** [package/feeds/packages/intltool/host/compile] Error 2
make[2]: Leaving directory `/home/build/openwrt'
make[1]: *** [/home/build/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/build/openwrt'
make: *** [world] Error 2
build@1c118a3b61d1:~/openwrt$
```

gmacario, 2015-01-12:
Solution: Install XML::Parser perl module

```
# apt-get -y install libxml-parser-perl
```

TODO: Add command in `easy-build/build-openwrt/Dockerfile`

closed 2015-01-13 by gmacario

Jean-Baptiste-Lasselle, 2018-11-20:
@gmacario thank you so much for that one, it was tricky ! I'm currently designing a dockerized pipeline based on tplink's fork of openwrt, and obviously they use versions prior to that merge, and you saved me days of work documenting that. Thanks to the whole openwrt community too !