gmacario/easy-build #198: Add baseimage implementing a GoCD Agent

gmacario, 2015-11-09:


gunnarx, 2015-11-16:
A useful idea - just make sure you check here:  
Looks like it's done already, or can easy_build improve upon it?  
https://github.com/gocd/gocd-docker

gmacario, 2016-01-19:
Opened a RFC Pull Request at https://github.com/gmacario/easy-build/pull/203

gmacario, 2016-01-20:
Merged PR #203 

Will close this issue as soon as someone can test it against go.genivi.org

gmacario, 2016-01-21:
Successfully tested go-agent inside Docker container

```
$ mkdir -p ~/test-goagent && cd ~/test-goagent
$ docker run -ti \
  --hostname=test-genivigo-agent1 \
  --user=go \
  --workdir=/var/lib/go-agent \
  --volume $PWD:/home/build \
  gmacario/build-yocto-genivi
```

Logged as go@container

```
$ vi /etc/default/go-agent
$ service go-agent start
```

CC @gunnarx

closed 2016-01-21 by gmacario