gmacario/easy-build #209: build-yocto-genivi: Add script to automatically build YGB 10.0.0

gmacario, 2016-01-26:
See http://lists.genivi.org/pipermail/genivi-meta-ivi/2016-January/000725.html

gunnarx, 2016-02-21:
You should find the steps for that already defined here:
http://go.genivi.org/go/admin/templates/meta-ivi-build/stages/Build/job/init_and_bitbake/tasks

Those tasks use a baseline-helper script to extract the layer hashes from the README and checkout the right commit. The materials will show you where to get that.
http://go.genivi.org/go/admin/pipelines/yocto-baseline-next/materials?current_tab=materials

If you have any problems accessing those links with your account, let me know because that's a bug.
And if you want to bring the helper script into easy-build, be my guest.

gmacario, 2016-02-22:
Thanks for the hints, @gunnarx!

> If you have any problems accessing those links with your account, let me know because that's a bug.

Yes I do - see related https://at.projects.genivi.org/jira/browse/TOOL-53

gunnarx, 2016-02-24:
Still working to understand the access control rules for templates.

Until that is solved, would you like to read some XML?  ;-)

The full build config is [backed up on Github](https://github.com/genivigo/server-config-backup) so you could search for the <templates> section and underneath that <pipeline name="meta-ivi-build">

gmacario, 2018-06-03:
Closing, see also #271

closed 2018-06-03 by gmacario