gmacario/easy-build #243: build-yocto-genivi: Failing when trying "bitbake nostromo-image"

gmacario, 2017-02-09:
I tried to build [meta-ivi N-0.1|https://at.projects.genivi.org/wiki/x/9AHe] with the following commands

```
mkdir -p ~/work
cd ~/work && docker run -ti --rm --name ygbbuild12 --hostname ygbbuild12 -v${PWD}:/home/build gmacario/build-yocto-genivi
```

then logged as build@container

```
mkdir $HOME/genivi-baseline
cd $HOME/genivi-baseline 
git clone https://github.com/GENIVI/meta-ivi
git clone git://git.yoctoproject.org/poky
git clone git://git.openembedded.org/meta-openembedded
cd meta-ivi; git checkout N-0.1
./scripts/checkout_layer_hash.sh poky
./scripts/checkout_layer_hash.sh meta-openembedded
cd -
export TEMPLATECONF=`pwd`/meta-ivi/meta-ivi/conf
source poky/oe-init-build-env build
MACHINE=qemux86-64 bitbake nostromo-image
```

The last command returns the following error:

```
build@ygbbuild12:~/genivi-baseline/build$ MACHINE=qemux86-64 bitbake nostromo-image
Please use a locale setting which supports utf-8.
Python can't change the filesystem locale after loading so we need a utf-8 when python starts or things won't work.
build@ygbbuild12:~/genivi-baseline/build$
```

gmacario, 2017-02-09:
Temporary workaround: Type the following commands before invoking `bitbake nostromo-image`

```
export LC_ALL=en_US.UTF-8
export LANG=en_US.UTF-8
export LANGUAGE=en_US.UTF-8
```

gunnarx, 2017-02-09:
I found this:
https://wiki.yoctoproject.org/wiki/TipsAndTricks/ResolvingLocaleIssues

gunnarx, 2017-08-24:
I was told about more people running into this problem with easy-build (GENIVI).  Can the **build-yocto-genivi** image include those environment settings?  Is it OK on **build-yocto** image?  (I guess it should happen for any morty-based yocto build, or something)

gunnarx, 2017-08-24:
Now that we know, the title of the bug might rather be something about the locale not being set on easy-build by default.

gmacario, 2017-09-07:
Closing issue, the root cause is tracked in #256

closed 2017-09-07 by gmacario