gmacario/easy-build #247: build-yocto-genivi: recipe dlt-daemon-2.16.0-r0: task do_configure: Failed

gmacario, 2017-03-23:
Failure when building meta-ivi 12.0 with [easy-jenkins](https://github.com/gmacario/easy-jenkins) and [my-genivi-pipelines](https://github.com/gmacario/my-genivi-pipelines/tree/build-meta-ivi-12.0-qemux86-64)

Excerpt from http://mv-linux-powerhorse.solarma.it:9080/blue/organizations/jenkins/my-genivi-pipelines/detail/build-meta-ivi-12.0-qemux86-64/2/pipeline

```
[ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ] Running shell script
+ id
uid=1000(build) gid=1000(build) groups=1000(build)
+ pwd
/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ
...
NOTE: recipe dlt-daemon-2.16.0-r0: task do_configure: Started
NOTE: recipe systemd-1_230+gitAUTOINC+3a74d4fc90-r0: task do_packagedata: Started
NOTE: Running task 1110 of 3421 (/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/meta/recipes-devtools/python/python_2.7.12.bb:do_packagedata)
NOTE: Running task 1228 of 3421 (/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/meta/recipes-multimedia/gstreamer/gstreamer1.0_1.8.3.bb:do_install)
NOTE: Running task 1254 of 3421 (/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/meta/recipes-connectivity/avahi/avahi_0.6.32.bb:do_compile)
NOTE: Running task 1428 of 3421 (/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/meta/recipes-graphics/drm/libdrm_2.4.70.bb:do_populate_sysroot)
NOTE: Running task 1473 of 3421 (/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/meta/recipes-extended/bash/bash_3.2.57.bb:do_install)
NOTE: recipe python-2.7.12-r1: task do_packagedata: Started
NOTE: recipe libdrm-2.4.70-r0: task do_populate_sysroot: Started
NOTE: recipe avahi-0.6.32-r0: task do_compile: Started
NOTE: recipe bash-3.2.57-r0: task do_install: Started
NOTE: recipe dbus-glib-0.106-r0: task do_install: Started
NOTE: recipe gstreamer1.0-1.8.3-r0: task do_install: Started
NOTE: recipe dbus-glib-0.106-r0: task do_install: Succeeded
NOTE: Running task 1474 of 3421 (/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/meta/recipes-core/dbus/dbus-glib_0.106.bb:do_populate_sysroot)
NOTE: recipe dbus-glib-0.106-r0: task do_populate_sysroot: Started
NOTE: recipe python-2.7.12-r1: task do_packagedata: Succeeded
NOTE: Running task 1475 of 3421 (/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/meta/recipes-core/libxml/libxml2_2.9.4.bb:do_package)
NOTE: recipe dbus-glib-0.106-r0: task do_populate_sysroot: Succeeded
NOTE: Running task 1476 of 3421 (/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/meta/recipes-connectivity/bluez5/bluez5_5.41.bb:do_configure)
NOTE: recipe bluez5-5.41-r0: task do_configure: Started
NOTE: recipe libdrm-2.4.70-r0: task do_populate_sysroot: Succeeded
NOTE: Running task 1477 of 3421 (/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/meta/recipes-gnome/gnome/gconf_3.2.6.bb:do_configure)
NOTE: recipe systemd-1_230+gitAUTOINC+3a74d4fc90-r0: task do_packagedata: Succeeded
NOTE: Running task 1478 of 3421 (/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/meta/recipes-core/dbus/dbus_1.10.10.bb:do_package)
NOTE: recipe gconf-3.2.6-r0: task do_configure: Started
NOTE: recipe libxml2-2.9.4-r0: task do_package: Started
NOTE: recipe dbus-1.10.10-r0: task do_package: Started
NOTE: recipe gstreamer1.0-1.8.3-r0: task do_install: Succeeded
NOTE: Running task 1479 of 3421 (/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/meta/recipes-multimedia/gstreamer/gstreamer1.0_1.8.3.bb:do_populate_sysroot)
NOTE: recipe gstreamer1.0-1.8.3-r0: task do_populate_sysroot: Started
NOTE: recipe gstreamer1.0-1.8.3-r0: task do_populate_sysroot: Succeeded
NOTE: Running task 1480 of 3421 (/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_1.8.3.bb:do_configure)
NOTE: recipe gstreamer1.0-plugins-base-1.8.3-r0: task do_configure: Started
NOTE: recipe bash-3.2.57-r0: task do_install: Succeeded
NOTE: Running task 1481 of 3421 (/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/meta/recipes-extended/bash/bash_3.2.57.bb:do_package)
NOTE: recipe bash-3.2.57-r0: task do_package: Started
ERROR: dlt-daemon-2.16.0-r0 do_configure: Function failed: do_configure (log file is located at /var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/build/tmp/work/core2-64-poky-linux/dlt-daemon/2.16.0-r0/temp/log.do_configure.686)
ERROR: Logfile of failure stored in: /var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/build/tmp/work/core2-64-poky-linux/dlt-daemon/2.16.0-r0/temp/log.do_configure.686
Log data follows:
| DEBUG: Executing python function sysroot_cleansstate
| DEBUG: Python function sysroot_cleansstate finished
| DEBUG: SITE files ['endian-little', 'common-linux', 'common-glibc', 'bit-64', 'x86_64-linux', 'common']
| DEBUG: Executing shell function autotools_preconfigure
| DEBUG: Shell function autotools_preconfigure finished
| DEBUG: Executing python function autotools_copy_aclocals
| DEBUG: Python function autotools_copy_aclocals finished
| DEBUG: SITE files ['endian-little', 'common-linux', 'common-glibc', 'bit-64', 'x86_64-linux', 'common']
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is GNU 6.2.0
| -- The CXX compiler identification is GNU 6.2.0
| -- Check for working C compiler: /var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/build/tmp/sysroots/x86_64-linux/usr/bin/x86_64-poky-linux/x86_64-poky-linux-gcc
| -- Check for working C compiler: /var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/build/tmp/sysroots/x86_64-linux/usr/bin/x86_64-poky-linux/x86_64-poky-linux-gcc -- works
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Check for working CXX compiler: /var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/build/tmp/sysroots/x86_64-linux/usr/bin/x86_64-poky-linux/x86_64-poky-linux-g++
| -- Check for working CXX compiler: /var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/build/tmp/sysroots/x86_64-linux/usr/bin/x86_64-poky-linux/x86_64-poky-linux-g++ -- works
[... 120 lines cut ...]
| 
| -- Configuring incomplete, errors occurred!
| See also "/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/build/tmp/work/core2-64-poky-linux/dlt-daemon/2.16.0-r0/build/CMakeFiles/CMakeOutput.log".
| See also "/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/build/tmp/work/core2-64-poky-linux/dlt-daemon/2.16.0-r0/build/CMakeFiles/CMakeError.log".
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/build/tmp/work/core2-64-poky-linux/dlt-daemon/2.16.0-r0/temp/log.do_configure.686)
NOTE: recipe dlt-daemon-2.16.0-r0: task do_configure: Failed
ERROR: Task (/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/../meta-ivi/meta-ivi/recipes-extended/dlt-daemon/dlt-daemon_2.16.0.bb:do_configure) failed with exit code '1'
NOTE: Running task 1482 of 3421 (/var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/meta/recipes-graphics/pango/pango_1.40.1.bb:do_package)
[... cut ...]
NOTE: recipe gstreamer1.0-plugins-base-1.8.3-r0: task do_configure: Succeeded
NOTE: Tasks Summary: Attempted 1482 tasks of which 1465 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /var/jenkins_home/workspace/ld-meta-ivi-12.0-qemux86-64-4SFCMADZTUA7EDU4P2OCJV6BHU2G2AIW4EOZISX2VMBVGKB32HPQ/poky/../meta-ivi/meta-ivi/recipes-extended/dlt-daemon/dlt-daemon_2.16.0.bb:do_configure
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
script returned exit code 1
```

gmacario, 2017-03-24:
Notice that the error log seems consistent to what currently triggered by https://go.genivi.org/

See https://go.genivi.org/go/tab/build/detail/yocto-baseline-next/8/Build/1/init_and_bitbake

gmacario, 2017-03-30:
It looks like an upstream bug - see discussion at http://lists.genivi.org/pipermail/genivi-meta-ivi/2017-March/001423.html

gmacario, 2017-04-25:
Please retry, meta-ivi gas introduced a workaround

https://github.com/GENIVI/meta-ivi/commit/6baf06d6f74f7a968e49ef6718bb8be6633ccb02

gmacario, 2017-04-25:
For the record, upstream bug is tracked at https://at.projects.genivi.org/jira/plugins/servlet/mobile#issue/DLT-4

gunnarx, 2017-08-24:
Looks like the bug was fixed.  Although it seems to have reappeared with the upgrade to pyro branch - (there's an open PR again) so maybe worth checking.

gmacario, 2018-06-03:
Closing, see also #271

closed 2018-06-03 by gmacario