gmacario/easy-build #251: build-yocto-genivi: Cannot build Docker image

gmacario, 2017-03-30:
Received the following error message from <mailto:no-reply@notify.docker.com>

> Hi Gianpaolo Macario,
>
> There seems to have been an issue with your Automated Build "gmacario/build-yocto-genivi" (VCS repository: gmacario/easy-build) during the build step.
You can find more information on
https://hub.docker.com/r/gmacario/build-yocto-genivi/builds/byztrkm8rwbh9kdhqgbbsfu/

Build failed: Error: image gocd/gocd-agent:16.1.0 not found

gmacario, 2017-03-30:
It looks like Docker image `gocd/gocd-agent:16.1.0` does no longer exists in Docker Hub.

Browsing https://hub.docker.com/u/gocd/ the following gocd agents are now available:

* gocd/gocd-agent-alpine-3.5
* gocd/gocd-agent-centos-6
* gocd/gocd-agent-centos-7
* gocd/gocd-agent-debian-7
* gocd/gocd-agent-debian-8
* gocd/gocd-agent-deprecated
* gocd/gocd-agent-ubuntu-12.04
* gocd/gocd-agent-ubuntu-14.04
* gocd/gocd-agent-ubuntu-16.04

Archie4Brasil, 2017-08-23:
Hello gmacario,
I am currently experience this error. How can I change the gocd-agent?

gmacario, 2017-08-24:
Hello @Archie4Brasil,

As a matter of fact I haven't been using this Docker image with https://go.genivi.org/ for quite a while.

Maybe you could instead try the scripts at https://github.com/genivigo/gocd-setup/

gmacario, 2017-08-24:
In any case I updated the Dockerfile to rebase on newer `gocd/gocd-agent-ubuntu-14.04:v17.8.0`.

See PR https://github.com/gmacario/easy-build/pull/258

closed 2017-08-24 by gmacario