gmacario/easy-build #265: How to change the configuration files?

naveen230, 2017-10-03:
I can't change the variables in local.conf and bblayers.conf files.I tried using vi vim and nano commands.Most importantly I wanted to add meta-mender (mender.io) layers and change the machine to Raspberrypi3.Is there a way to do that?

Thanks

naveen230, 2017-10-04:
the solution is to install vim in the docker container
sudo apt-get update
sudo apt-get install vim

gmacario, 2018-06-03:
Hello @naveen230 ,

Not sure which Docker image of the easy-build project you are referring to.

The general idea is that the image should not contain local.conf and bblayers.conf which may be mounted from a volume, or even reside on the Docker host, therefore they can be edited with any editors installed on the host itself.

Honestly I do not see the need for installing an editor inside the container.

Please correct me if I am wrong

gmacario, 2018-08-19:
No feedbacks received, closing

closed 2018-08-19 by gmacario