gmacario/easy-build #285: Network proxy error when I build Yocto source of E3900-MR3.1 for APL.

wt-kevin, 2019-12-01:
After finishing the settings, I ran ./setup.sh on VMWare with Ubuntu 14.04, but encountered network proxies problem as below,
Duplication steps：
kevin@ubuntu:/Yocto/iotg-yocto-bsp-public$ ./setup.sh
Select an option: 
1. Build kernel image with CAVS HD Audio driver (Default)
2. Build kernel image with CAVS SSP Audio driver
Default option is build kernel image with CAVS HD Audio driver. If no input is received within 20 secs, default will be used.
1
Select an option: 
1. core-image-sato-sdk (Default)
2. core-image-sato
3. linux-kernel
4. custom
Default build target is core-image-sato-sdk. If no input is received within 20 secs, default target will be built.
1
--2019-11-30 16:22:39--  https://raw.githubusercontent.com/openembedded/openembedded-core/master/meta/conf/layer.conf
Connecting to 192.168.40.2:8081... failed: Connection refused.
ERROR: wget fail to fetch https://raw.githubusercontent.com/openembedded/openembedded-core/master/meta/conf/layer.conf
NOTE:
You are either not connected or behind corporate proxy
Please make sure network proxies are setup at /etc/environment or
~/.bashrc :-
http_proxy=<http proxy server path>:<port>
ftp_proxy=<ftp proxy server path>:<port>
https_proxy=<https proxy server path>:<port>
socks_server=<SOCKS server path>:<port>
kevin@ubuntu:/Yocto/iotg-yocto-bsp-public$

gmacario, 2019-12-11:
Hello,

Sorry for my late reply.

My environment does not need http proxy settings so I have never encountered the problem you mentioned.

At any rate, have you tried to define the environment variables as suggested in the error message above?

Also, did you try to invoke "wget https://raw.githubusercontent.com/openembedded/openembedded-core/master/meta/conf/layer.conf" directly on the command line? If so, which is the result you got?

wt-kevin, 2019-12-11:
Hi gmacario,
I fix this problem, and other build logs are found(log is as follows), could you help on it? thank you.
[Yocto_Build_Error_Log.log](https://github.com/gmacario/easy-build/files/3949672/Yocto_Build_Error_Log.log)

gmacario, 2019-12-15:
Hello,

By looking at the log you provided it looks like you have a problem on your specific Yocto build.
I do not think this is an easy-build issue.

Closing issue as I have no experiences with this particular target you are building for.

closed 2019-12-15 by gmacario

wt-kevin, 2019-12-16:
Hi  gmacario,
Thank you, Could you let me know how we can issue Yocto build problem?

gmacario, 2019-12-16:
Hi @wt-kevin,

>Could you let me know how we can issue Yocto build problem?

Please refer to the documentation of your custom BSP layer ("E3900-MR3.1 on APL" according to the description of this issue).

wt-kevin, 2019-12-18:
Hi gmacario,

I googled it and can't find it, could you let me know the file name? or link to download? thank you.