gmacario/easy-build #287: I/o error when building in shared directory (windows host)

bok11, 2020-05-18:
I am experienceing two errors when trying to build, first i get an error with IO:

```Python
ERROR: Build of do_configure failed
ERROR: Traceback (most recent call last):
  File "/workspaces/poky/bitbake/lib/bb/build.py", line 560, in exec_task
    return _exec_task(fn, task, d, quieterr)
  File "/workspaces/poky/bitbake/lib/bb/build.py", line 500, in _exec_task
    exec_func(func, localdata)
  File "/workspaces/poky/bitbake/lib/bb/build.py", line 229, in exec_func
    exec_func_shell(func, d, runfile, cwd=adir)
  File "/workspaces/poky/bitbake/lib/bb/build.py", line 365, in exec_func_shell
    os.mkfifo(fifopath)
OSError: [Errno 5] Input/output error
```

I also seen a Socket error, but cannot reporudce taht on this machine (can on my home machine though)

To reproduce, do the following from a windows host:

```bash
docker run -v <local folder>:/workspaces -it gmacario/build-yocto
cd /workspaces
git clone -b jethro git://git.yoctoproject.org/poky
cd poky
source oe-init-build-env build-test
bitbake -k core-image-minimal
```

If i save the files locally in the docker container, i do not see this issue (i run out of space though)

gmacario, 2020-05-18:
Hello,

Thanks for reporting.

As a matter of fact I have never used easy-build on Windows hosts.

Anyway by reading the errors in your logs it looks like the directory which you mounted from the Windows host has some limitations with respect to a native Linux filesystem.

I would suggest to add some debugging messages to file /workspaces/poky/bitbake/lib/bb/build.py

gmacario, 2021-07-07:
No follow-ups received so far, closing issue

closed 2021-07-07 by gmacario